SolarTherm/SolarTherm #10: Complete setup guide for Ubuntu 14.04

paul-scott, 2015-11-09:


closed 2015-11-11 by paul-scott

jdpipe, 2015-11-11:
just out of curiosity, other bug-trackers I've used have had a feature where the bug is linked to a changeset/revision where the solution was implemented/completed. Anything like that in github?

paul-scott, 2015-11-11:
@jdpipe Yep you can mention them inline by just copying the hash of the commit directly into the comment.

For example: "fcbe10c5fb8e0f62bed9896335d97615718beeda" gets formatted into fcbe10c5fb8e0f62bed9896335d97615718beeda

I'll try to do this from now on when closing the issues.  The other option is to reference the issue directly in the commit message.  For example using "Fixed #10 blah blah" in the  commit it will automatically reference and close the issue.