SolarTherm/SolarTherm #13: enable testing/running without needing root privileges

jdpipe, 2015-11-10:
It should be possible to build, test and run solartherm without needing to be root.

I mostly managed to achieve this by

cmake -DCMAKE_INSTALL_PREFIX=~/local

but the part where it files is the calls to distutils in the src/python dir. I think this can be fixed by somehow setting distutils.sysconfig.PREFIX, or passing 'prefix' to the setup.py in another way.

Then, non-root tests could be run after setting

OPENMODELICALIBRARY
LD_LIBRARY_PATH
PATH
PYTHONPATH

Potentially we could use a driver script 'st' which could set those variables automatically, depending on the location of the 'st' script...?

paul-scott, 2015-11-11:
Yep I'm starting to work on this now.

I'm not sure if ~/.local is meant for this purpose or not?  It might better to create a new directory with bin and lib subdirectories.  Then just have to update PATH and LD_LIBRARY_PATH so that they point there.

We can easily use the standard ~/.openmodelica/libraries/ folder for modelica libraries without having to setup any env variable.

Maybe use virtualenv for the python stuff (working out how to use it now).

jdpipe, 2015-11-11:
My ~/.local idea is basically to allow one single simple change between 'user' installation and 'root' installation (via CMAKE_INSTALL_PREFIX). It's the easiest way forward. I am extrapolating a bit from the fact that ~/.local/share is for user-specific data files https://www.reddit.com/r/linuxquestions/comments/3krb06/where_did_homeuserlocalshare_come_from/ hence ~/.local/bin is for user-specific binaries, right??

Options like virtualenv, ~/.openmodelica, etc, are a bit more complex, since we would still want to use default 'root' locations for the distributable packages. But they could have some advantages.

Mostly I am thinking that the 'user' installation is for developers of SolarTherm, but also for users working on locked-down systems with no root access.

Another option is to allow everything to be run in-place, in the ~/solartherm (or whatever) source/build tree. This could also be done quite nicely, using a little bash/python script to set up the environment. Actually even more convenient for developers, I would say.

paul-scott, 2015-11-11:
"hence ~/.local/bin is for user-specific binaries, right??"  <- this is the bit I'm not sure about.  This would seem like a sensible convention to me, but I couldn't find a reference to anyone putting anything in there but share.

In relation to a script for setting up variables, do you see this as a permanent modification to something like .bahrc or a temporary environment that gets pulled up and down?  virtualenv already has this functionality that we could leverage, the only thing it appears to be missing is that it doesn't set up the library path.

So if you want a local installation the process would be something like:

**Install any system-wide packages**

virtualenv --system-site-packages <install directory>
source <install directory>/bin/activate

**install any optional python dependencies locally**
**install solartherm with -DCMAKE_INSTALL_PREFIX=<install directory>**

deactivate

This would require an activate call in each terminal before using the project.

paul-scott, 2015-11-11:
I've included instructions in the documentation on how to get a local installation up and running.  It ain't pretty but it works for now.  I also have the travis-ci builds + tests passing!

paul-scott, 2015-11-12:
Closing for now as the capability was added in ee74878f190829bc4779e95955111b238ec2ee66.  We might reopen this or create a new issue later on to improve the solution.

jdpipe, 2015-11-12:
I made a additional tweak, to allow the use-case without virtualenv, see 918af793b5bc4e0ed59eecc60e248d05b1d608cb. First commit to this repo, hope it works :-)

paul-scott, 2015-11-12:
OK cool.  So in order to do things this way you would need to set up PYTHONPATH right?  The other thing they would need to do is set up LD_LIBRARY_PATH and LIBRARY_PATH for any C libraries.

jdpipe, 2015-11-12:
Yes, this approach requires all those messy environment vars. Interestingly, I found I only needed to use
export LD_LIBRARY_PATH=~/.local/lib
export PATH=$PATH:~/.local/bin
export OPENMODELICALIBRARY=~/.local/lib/omlibrary:/usr/lib/omlibrary
I was surprised that I didn't need to use PYTHONPATH...

paul-scott, 2015-11-12:
Weird, maybe something is hanging around from previous install?  I had trouble with OPENMODELICALIBRARY before (long time ago) so have tried to steer away from using it.

I just realised the change you made will likely break certain root installs.  For example, on my system python is under /usr/, but I have installed openmodelica and SolarTherm stuff under /usr/local (using prefix).

Also if you don't set LIBRARY_PATH then one of the tests should fail when it tries to compile the simulation executable (unless libsolartherm_test.so is hanging around somewhere else).

jdpipe, 2015-11-12:
The PYTHONPATH thing looks like a 'feature' of some sort...?

john@novo:~/solartherm$ env |grep "\\.local"
LD_LIBRARY_PATH=/home/john/.local/lib
OPENMODELICALIBRARY=/home/john/.local/lib/omlibrary:/usr/lib/omlibrary
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/john/.local/bin
john@novo:~/solartherm$ python 
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> print sys.path
['', '/usr/lib/python2.7', '/usr/lib/python2.7/plat-x86_64-linux-gnu', '/usr/lib/python2.7/lib-tk', '/usr/lib/python2.7/lib-old', '/usr/lib/python2.7/lib-dynload', '/home/john/.local/lib/python2.7/site-packages', '/usr/local/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages', '/usr/lib/python2.7/dist-packages/PILcompat', '/usr/lib/python2.7/dist-packages/gtk-2.0', '/usr/lib/pymodules/python2.7', '/usr/lib/python2.7/dist-packages/ubuntu-sso-client']
>>> 

You can see that sys.path here includes .local/lib/python2.7, even though nothing in my env tells python to look there.

Personally I don't think that 'make install' should be installing anything in /usr if the user has requested PREFIX=/usr/local. Instead, env should be set include /usr/local/lib/python2.7 in the python path (which from above may actually happen automatically).

jdpipe, 2015-11-12:
As for LIBRARY_PATH... I am not familiar with that environment variable. When does that come into play, exactly?

jdpipe, 2015-11-12:
Perhaps need for LIBRARY_PATH is avoided by -L[dir] flags in the gcc command line?

paul-scott, 2015-11-12:
LD_LIBRARY_PATH is for ld linking during when running an executable, LIBRARY_PATH is for gcc/clang to link during compilation.  I don't fully understand why both are needed but I know I've needed to use it in the past when things were installed in funny places, and for me it fails if I don't set it up.

That is odd I'm not getting this behaviour in python: if I install it under /usr/local/ then the python paths aren't automatically setup.

Anyway for now I added a variable so that I can turn off forcing the prefix for pip (-DFORCE_PYTHON_PREFIX=0).  By the default it has the behaviour you implemented.

jdpipe, 2015-11-12:
If you want to catch an ubuntu, you have to think like an ubuntu :-) In /usr/lib/python2.7/site.py, I found a function "getusersitepackages" which explicitly adds ~/.local/lib/python2.7/site-packages", if it is found to exist. Also found https://leemendelowitz.github.io/blog/how-does-python-find-packages.html and comments in the site.py file mentioned which clearly shows that Debian/Ubuntu have got custom behaviour here.

jdpipe, 2015-11-12:
Regarding LIBRARY_PATH, I always use -L to specify any unusual library locations. I am fairly sure that will be equivalent.

paul-scott, 2015-11-12:
OK that explains it :)

Yeah I'm pretty sure, but the makefile for compiling the simulator gets generated automatically by omc so it might not have the right flags set.

jdpipe, 2015-11-12:
Again on LIBRARY_PATH, am I understanding correctly that currently we have `TestExternalC.c` in `src/c` which is currently only there as a proof-of-concept of how we can call external C code from inside our Modelica code, even though at present we're not using it.

I noticed that my `SimpleSystem.makefile` already automatically includes a bunch of subfolders in `~/.local/lib/omlibray` all by itself. So I guess something is already happening in the `omc` command based on the value of `OPENMODELICALIBRARY` where `libsolartherm_test.so' has already been installed.

paul-scott, 2015-11-12:
Yep exactly.  So it doesn't matter if we don't get it sorted at this stage.

It looks like wherever it ends up finding SolarTherm it then includes all these subdirectories.  It probably makes sense to put any C libraries we have under SolarTherm/Resources/Library/.

jdpipe, 2015-11-12:
Yes, if they are expected to be installed on the OPENMODELICALIBRARY path, then the source code might as well be in our tree where other modelica library files are stored...?

paul-scott, 2015-11-20:
Have simplified the build and install instructions a8ca35234e5064646f7d87d1debc3db58f25ca82.  The local build instructions no longer require virtualenv, although travis-ci continues to use it for now.

The default behaviour is to install everything under the provided prefix.  If required, environment variables can be automatically set for the current terminal by sourcing build/src/env/st_local_env.  Otherwise if the EVs can be made permanent by just copying them from this script into ~/.bashrc.

closed 2015-11-20 by paul-scott