SolarTherm/SolarTherm #14: Units of measurement on st_simulation, st_plotmat cmdlines

jdpipe, 2015-11-12:
Perhaps it would be useful to have some option of specifying suffixes on the numerical command-line parameters like --stop, --step, --xlim and also then perhaps on the parameter value options like t_storage, P_rate, etc? Even if we could just use h,d,y if would be a big help.

st_simulate --stop 1y --step 300s FluidSystem.mo t_storage=6h P_rate=75kW
st_plotmat --xlim 10d 17d SimpleSystem_res.mat wea.wbus.dni Q_flow_chg,Q_flow_dis

Trick would be to keep this simple -- perhaps a simple dict containing the value of the suffixes, and then some lists containing the permissible suffixes for different command-line parameters?

paul-scott, 2015-11-12:
We have similar thoughts, I already added this feature earlier today and updated the examples in the tutorial cc018cbb55f1007c0a9b42e2906c17ac6ac565c1.

jdpipe, 2015-11-12:
:-) great!

paul-scott, 2015-11-12:
Ok reading this again, I have done it for the time values but not the parameter value quantities.  I see what makes sense for the latter.

paul-scott, 2015-11-20:
New unit conversions can be added to unit_conv dictionary in simunation.py

commit 7f4fd114 (2015-11-20): Fixed #14 - added parameter unit passing functionality
closed 2015-11-20 by paul-scott