SolarTherm/SolarTherm #15: Question: LCOE calculation result when st_simulate run for partial year?

jdpipe, 2015-11-12:
I found that if I run st_simulate for less than a whole year, eg 
```st_simulate --stop 3153600 --step 300 FluidSystem.mo t_storage=6 P_rate=75000```
instead of
``` st_simulate --stop 31536000 --step 300 FluidSystem.mo t_storage=6 P_rate=75000```
then I still get a reported LCOE, of 97 AUD/MWh instead of 159 AUD/MWh. To avoid LCOE data being mis-quoted, I suggest that the LCOE only be reported to the user in the case where a full-year simulation has been completed (or at least a warning should be given).

paul-scott, 2015-11-12:
What it does is calculate the LCOE based on the average energy per year obtained from the sim.  So if you only run in summer months if will underestimate the LCOE, and if you only run in winter months it will overestimate the LCOE.  This also means if you run for a year and a half it will give misleading data (currently weather data only lasts a single year anyway).  I think there should be a warning and maybe also drop the data.

jdpipe, 2015-11-12:
I think it makes sense just to drop the data (ie make no output, i presume?) and say that LCOE can only be calculated with a full-year simulation. Anything else is interesting, but rather misleading (especially if you're not paying a full year of loan repayments)

closed 2015-11-20 by paul-scott