SolarTherm/SolarTherm #16: receiver thrashing!

jdpipe, 2015-11-14:
When simulating a typical simple scenario,

st_simulate --stop 1y --step 1m SimpleSystem.mo t_storage=6 P_rate=75000
st_plotmat --xlim 5d 10d SimpleSystem_res.mat wea.wbus.dni:P_elec E:Q_flow_chg,Q_flow_dis

I see a lot of on-off switching in the receiver in this data. What we've implemented here is something that trips off the receiver, and keeps it off for some time. Although I think this gives us reasonable results as far as system output goes, I think it's not what a real system would do, and people looking at our graphs will query our approach.

I think a better approach in this case would be partially off-steer the heliostats, so that power delivered is equal to the power currently being consumed by the power block, hence keeping the storage full or nearly full. Perhaps we could discuss a change to the control strategy that would implement this. It may have the benefit of speeding up the simulation, as less events will be triggered during a sunny day.

paul-scott, 2015-11-25:
I've created a concentrator model that can be off-steered 063c71dd98b3939565e02460171cc0149a6749e5.  You can specify a rate of change which indicates what fraction of mirrors can move on/off sun per second.

The other thing that would be good to do is to work out a proper way of turning the receiver "off".  Currently I have a boolean variable which supposedly shuts a "door" on the receiver cavity, preventing any losses.  In practice is it the case that receivers have some form of recirculation from the hot tank that is used to prevent the HTF from solidifying?  It is probably worth modelling this, for accuracy and because it will enable me to remove the "door".

Thoughts?

jdpipe, 2015-11-25:
Some receivers are fully drained when the go off-sun, eg to stop salt from solidfying in them. Others have HTF circulated through them constantly (even at night), in some cases with a cover swinging into place. Others, with fluids like air/water are just allowed to go cold.

The key thing here is that you can indirectly set the offsteer fraction to maintain storage at 100% (ie when the storage is full, the receiver power reduces to what the power block is requesting, since no more can be stored right not). Is that possible now?

paul-scott, 2015-11-25:
So far it is just one piece.  We still need to develop the control.  The controller can read things like the level of the hot water tank, the sun intensity, the receiver mass flow rate and the power block mass flow rate.  I'll try something out but I'm open to ideas.