SolarTherm/SolarTherm #17: include construction time in LCOE calc

jdpipe, 2015-11-19:
Not sure if we do this yet... can we include the number of years of construction time in the LCOE analysis? 

```python
#!/usr/bin/env python
# John Pye, 19 Nov 2015

# numbers are broadly based on Gemasolar

# capital cost, EUR
C = 230e6

# number of years before operation begins
# capital is spent at a constant rate through this time
# must be an integer, I think.
s = 2

# annual energy generated, kWh
E = 110e6

# annual O&M cost, same units as C
O = 0.018 * C

# number of years of operation
n = 25

# compounding rate (1 + interest rate)
# wouldn't be too hard to expand this to a weighted average cost of capital
r = 1.08

# amortisation factor
R = r**n * (r**s - 1)/(r**n - 1)

# LCOE, if all capital spent at t=0, O&M costs spent at t=1,2,...n
L = C*R/(s*E) + O/E

# let's calculate it another way to check
num = 0
den = 0
for i in range(n+s):
	if i < s:
		num += (C/s)/r**i
	else:
		num += O/r**i
		den += E/r**i
L2 = num / den

for s in "C,O,n,r,R,L,L2".split(","):
	print "%s = %f"%(s,eval(s))
```

jdpipe, 2015-11-19:
As a bonus, perhaps we can consider the time of year when loan commences, and time of year when operation commences, and examine the affect of sliding the whole project back/forth by fractions of a year (and effect of fractional years for total construction time).

paul-scott, 2015-11-19:
Yeah I don't see why not.  So the only extra thing we need is the number of years of construction right?  We could put this with the rest of the parameters in the model if you think that makes sense?

I'll get you to check what I end up doing.

paul-scott, 2015-11-19:
It might make sense to build in an override for financial parameters like this.  This way you redo the calculations without having to rerun the simulation each time.

paul-scott, 2015-11-19:
I'm not sure how we should handle fractional years.  Does this mean we should move to a monthly compound rate instead of yearly?  This is the only way I could see how it would make a difference...

jdpipe, 2015-11-20:
if we complete construction in 2.5 y instead of 2 y, it could mean one extra summer of production early in the project, which would make quite a difference. also, it means that the loan starts getting paid off six months earlier, which would also make quite a different. I supposed that monthly calculations would be the easiest way to see this effect. Over the life of the plant, we just pluck out the months from the annual performance calculation, in a loop.

Not urgent, something to hold for later, probably.

paul-scott, 2015-11-20:
Have a solution in c5820ad86bf975cbc5e89ab80b84094254ae978e.  New issues have been created for the other things discussed here.

closed 2015-11-20 by paul-scott