SolarTherm/SolarTherm #22: Modelling cocentrator tracking flux

paul-scott, 2015-11-24:
The current design allows receivers to have multiple "elements", which are flat elements that can receive flux on one face.  We need a concentrator design that maps from the solar conditions to the flux on each of these elements.

The core component needs to map from alt, azi to a vector of fluxes (per incident irradiance) which represents the radiance hitting the receiver elements under normal tracking conditions.  One approach is to have a table of points in this space, and then interpolate between them.  This table could be provided as a file, or it could be calculated during modelica initialisation, where certain parameters such as table granularity can be provided.  Ultimately this could be calling something like Tonatiuh as part of this initialisation.

A quick hack to get this working would be to make use of CombiTable2D in the MSL.  It is a bit messy as it will require `nelem` tables, so some of the calculations will be repeated multiple times.  Ultimately we will likely want our own C code to do the interpolation, that can read from a file and/or interface with another tool that performs the flux mapping.

jdpipe, 2015-11-24:
Modelling receiver dynamics is complex; it involves all the hydrodynamics within the receiver at least as much as the flux profile. Modelling the flux profile isn't very helpful unless we start also modelling 'on-steer' and 'off-steer' processes, which I don't see us doing yet for quite a while.

For the moment, I suggest that we need a 2D lookup table of integrated *field+receiver* efficiency. I think it should be stored as data in solar-polar coordinates, so hour angle vs declination angle, for a typical location. Open to discussion on this. Victor and Manuel (CSIRO) state that as few as 32 points is enough for a ±1% accurate field performance estimate year-round. We should plan on being able to use their data, or finer-res data if it is available. Normally we will be calculating the performance specifically for location of interest, so I think the hour-vs-declination data table is appropriate.

Longer term, a subdivided receiver may be desirable, but note that the flux on each element will be a complicated function of the system state, and not just the time; in that case it won't just be a simple lookup.

paul-scott, 2015-11-24:
I was trying to do something similar to what it appears SAM is doing so we can do a comparison.  From what I can tell it cuts the receiver up into panels and calculates the flux on these panels for a bunch of sun positions before running the simulation.  The plan was to have the flux lookup for the tracking position, and then use some simple rules to modify this flux based based on the system state (off-sun, moving onto sun, offsteer some mirrors).

We can do the integrated field+receiver, but won't it be an even more complicated function of the the system state?  Things that will have an impact which I imagine we would have to ignore with a more lumped approach:

- heliostat positions
- ambient temperature
- wind
- HTF flow rate
- HTF inlet temperature

jdpipe, 2015-11-24:
Good points, Paul. I guess a lumped field+receiver could work well for the optics, but it can't work for the whole problem, because of the variables you mention of ambient temperature, wind, HTF flow rate, HTF inlet temperature.

So perhaps we need field+receiver optical model, which will give us receiver absorbed irradiance as a function of sun position.

Then we need a receiver thermal model, which can give us receiver losses as a function of sun position, ambient temperature, wind, HTF flow rate, HTF inlet temperature. Be careful about assumptions of 'input parameters' and 'output parameters'; the definition is not strong with Modelica, and you can create 'index problems' depending on how you phrase component models (I believe). Not sure if you have dug into this topic yet or not.

The field+receiver optical model should hopefully also give us some extra information like the attenuation losses, spillage losses, mirror reflectivity losses, blockage and shading losses, cosine losses. But really the only one that matters for the annual performance model is the absorbed irradiance.

The receiver thermal model needs to be a function of sun position too, because the incident flux affects external wall temperature profiles (eg for cavities), and hence the losses. But we can give the user the option of ignoring some input parameters if they don't want to use them, which will allow us to have receiver models of varying complexity.

What do you think about this... this approach allows the implementer to decide how much resolution/detail then need, but keeps the high-level view simple.

On top of these two things, we also need to consider some extra detail for the operating strategy/switching states for the receiver. Maybe we want to add some extra physics to describe what happens during on-steer, off-steer, partial off-steer (dumping), and maybe other things.

paul-scott, 2016-02-23:
This issue was duplicated and continued in #24, which is now resolved.  Should come back to the points here as the approach evolves.

closed 2016-02-23 by paul-scott