SolarTherm/SolarTherm #26: Weather data insolation offset

paul-scott, 2016-02-04:
DNI from weather file doesn't match with calculated alt and azi.

This is likely because from memory the TMY data provides the DNI and GHI for the preceding hour.  This creates a lag in the insolation.  Other weather data is reported instantaneously?

The other option is that the `tstart` the local time at t=0 and the start of the year used by solar position calculator is wrong.

jdpipe, 2016-02-04:
Correct, you will see DNI before/after sunset, and that is one reason for maybe hacking your own interpolation routine for DNI instead of using the default one. You could apply constraints to keep instantaneous DNI between the clear-sky maximum value and zero, knowing the sunrise/sunset times.

You might like to refer also to 
http://dx.doi.org/10.1016/S0038-092X(01)00052-4
http://ascend4.org/Data_reader#Further_Work

paul-scott, 2016-02-16:
In f81ea055a0346535466bcbb9edc5e1b46ba9990a added ability to set the time delay of a weather file column.  For example, we could delay DNI and GHI by 30 minutes.  However, we found that it appears to depend on the weather file whether or not this actually lines up better.  For example, in the ``AUS_NT.Alice.Springs.Airport.943260_RMY`` file the DNI and sun altitude already line up very well, whereas the ``USA_CA_Daggett.Barstow-Daggett.AP.723815_TMY3`` file benefits from a 30 minute delay.  By default the delays are set to zero for all columns.  These can be fine tuned for particular weather files if desired.

Even if we improve the interpolation approach to take into account the sunrise/sunset times and preserve the mean, at some point we would still have to problem of having to deal with the fact that the instantaneous weather measurements and cumulative irradiance measurements are in a sense out of sync.

jdpipe, 2016-02-16:
Probably worth reviewing the documentation for particular datasets and/or file formats. Perhaps there are different conventions, eg data is at 10:00 am is the average from 9:30--10:30, or is the average from 10:00--11:00, etc.

paul-scott, 2016-02-24:
Looking at results output form SAM, it appears to treat both the instantaneous and integrated results as occurring at the same time instant.  They directly copy the TMY3 data from 01:00-24:00 range to 00:00-23:00 range, moving it back in time by an hour (can be seen by comparing NREL TMY files to SAM TMY files).  It appears that the sun positions are calculated for the first point on the half hour: 00:30.

What we currently do is treat the first time point as occurring at 01:00 and calculate the sun positions for this time also.  If we leave all delays off, but change the start time from the beginning of year in the motab file from 1 hour to 30 minutes (3600 to 1800) we get a match with SAM (when aligning first points of output at zero seconds).

Not sure how we should progress in this regard.

jdpipe, 2016-02-24:
What would happen if we didn't use interpolation, so we have a step change in DNI (etc) at hourly increments, and also fix the sun position to the same spot for that entire hour? This is like a quasi-steady model for each hour, with only the storage dynamics changing within the hour. Once we had that, we could then implement interpolation and find out which smoothing gives reasonable accuracy.

Another approach would be to take short time-step TMY data (I think I can find some of that for you) and the to average it ourselves, and observe what approach to interpolation best preserves the overall results when resampled with different timestep lengths...

jdpipe, 2016-10-19:
Perhaps we could look into this in the context of **clear sky DNI** There is a well-understood formulation for the 'envelope' of maximum DNI on a particular day owing to turbidity. Based on this we can avoid incorrect sampling of TMY time points. Just a thought.