SolarTherm/SolarTherm #34: Find source of error in TroughWagner Daggett solar positions

paul-scott, 2016-02-12:
When using the Daggett weather data, the solar positions go outside bounds of acos calculation.

jdpipe, 2016-02-12:
Sounds like a problem to do with the sign convention in your sun position calculator. Probably the sign convention should be different depending on the hemisphere to avoid unnecessary discontinuities from arc-trig functions.

paul-scott, 2016-02-15:
Works perfectly with same time steps in a separate model.  Seems to be a slightly different numerical values when used in with the generic model.  In practice hard to test for this kind of thing as unit testing the solar positions in isolation won't show this error.

In the TroughWagner model the acos argument takes on a value of -1.000000000000000444 when it needs to be in interval [-1, 1].  Will try calling the trusty old max to make sure this remains in interval.  Should only trigger a couple of times a day.

commit 625eabca (2016-02-15): Fixed #34: applied max and min to solar position acos arguments
closed 2016-02-15 by paul-scott