SolarTherm/SolarTherm #35: Get it running under Windows

paul-scott, 2016-02-14:


jdpipe, 2016-02-15:
I would suggest trying to use the relatively new MSYS2 platform for this. It is a nice GNU-based environment with a decent package manager, and I know it includes Python support and some GUI stuff.

paul-scott, 2016-02-15:
Sure I'll give that a shot.

Technically it should be possible to run everything natively on windows, but in practice it will be a pain.  As a short-mid term compromise I think MSYS2 sounds good.

paul-scott, 2016-03-18:
I've been trying this on and off over the last couple of days.  Just wanted to document some of my thoughts and see if anyone has any comments.

- OpenModelica is compiled with as a 32-bit application using mingw infrastructure.  The whole model compilation process also uses this.
- Ignoring the st scripts for now, in order to get the basics working under OMEdit, we just need to work out a solution for the external C code (e.g., the optical efficiency interpolation).  There are 3 options:
  - Compiling a shared library.  In windows this is more difficult, because as opposed to linux, dynamic libraries need to have all symbols defined at compile time.  This means that using ModelicaError requires us to link against all the OpenModelica runtime!
  - Compiling a static library.  If we do this under MSYS2, there is a problem in that the mingw libraries are not compatible with the OpenModelica mingw libraries (old version of gcc and mingw).
  - Providing an include file that gets compiled with the model.  Here the problem is that we don't have control over how our code is compiled.  On the build server this caused a problem because the gcc is old and OpenModelica tries to compile it in a way that is not compatible with c99.  Forcing OpenModelica to use c99 causes its own problems...

For the first approach I need to work out exactly what additional libraries I need to link to.  I could take the second approach and try to compile it using the OpenModelica mingw, but I'm not sure how to go about this.

Will continue to experiment.

paul-scott, 2016-03-18:
OK couldn't get either of the above two approaches to work.  Although it is not ideal, what will work is to take the third approach, and make sure my code is compatible with old gcc versions (for windows OM and older ubuntu) without requiring any additional compiler flags.  Practically what this means is that I just have to go through the solinterp code and remove any variable declarations in for loops.

This is the most practical solution I can come up with...

jdpipe, 2016-03-18:
I never had too many difficulties with DLLs on Windows in my work with ASCEND project (http://ascend4.org/Writing_ASCEND_external_relations_in_C). There were some details about symbol import/export to be attended to, but otherwise straightforward. Not sure if any of that helps at all...?

paul-scott, 2016-03-20:
Yes I think the main reason is because you would be building ASCEND and the external C in the same mingw environment?  We could attempt to build OM under MSYS2, but I imagine this would be a nightmare, or we could try to tap into the OM mingw, but I don't know how to do this easily.

For now I have taken the third option listed above, and implemented the changes in: 92ed140112e87dda79348035405b4bae2ad27001.  If we have more complicated external C build requirements then this will need to be revisited.

So what this means is that the SolarTherm library can be run from within OMEdit on windows by just loading the library from the repository source.  Getting the python scripts installed and running smoothly is the next task.  I've had some of them running, but I haven't decided on the easiest way yet.

jdpipe, 2016-03-21:
So you moved to ANSI-C instead of C99; does that mean we're all good, both on Windows as well as with the build/test server?

paul-scott, 2016-03-21:
Yup.  The SolarTherm library is all working through OMEdit (apart from its use of st_wea_to_mo when weather files are in the wrong format, but we might want to change this anyway).  The other scripts are not working in windows.  I can get the python library working, but the environment isn't set up to actually build the simulation properly.  Again this is because a different mingw was used for OM, and I can't work out how to access this environment...  

Maybe next time we meet I can show you the problem and your extra experience with mingw/msys might find a solution.

jdpipe, 2021-01-06:
We still have some work to do with the external functions, but the Python scripts are now updated to Python 3 and working on Windows. See https://github.com/SolarTherm/SolarTherm/wiki/Running-SolarTherm-on-Windows-%28MSYS2%29.