SolarTherm/SolarTherm #36: Round 2 of financial calcs

paul-scott, 2016-02-16:
What additional financial/performance calculations do we want?  Should we prioritise some of these over the short term?

- depreciation
- yearly performance degradation
- PPA
- loans
- taxes
- incentives
- NPV
- IRR
- etc...

jdpipe, 2016-02-16:
For now, I think a simple LCOE with no consideration of taxes etc is appropriate. LCOE is a metric to evaluate the costs independent of project finance, economic context, etc. LCOE is affected by the magnitude of O&M and capital costs, together with annual output and discount rate. Nice and simple.

More complex models bring in the things you mentioned. I don't see them as being short-term needs.

A nice reference scenario for LCOE across various technologies: http://www.co2crc.com.au/dls/Reports/LCOE_Report_final_web.pdf

paul-scott, 2016-02-23:
Low priority for now so closing.

closed 2016-02-23 by paul-scott