SolarTherm/SolarTherm #42: Add to st_plotmat the ability to also load in csv for comparison

paul-scott, 2016-02-23:


jdpipe, 2016-02-24:
good idea, i take it we can pull out CSV data eg from SAM? we could include some datasets in our code repo as part of test cases, too.

paul-scott, 2016-02-24:
Yea, that was the original reason why I wanted it.  Can also compare two sets of results from SolarTherm (whether they are mat or csv files).  I've got an implementation working which will close this in an upcoming commit.  Example use is:

``st_plotmat TroughWagner_res_0.mat:sam_wagner_results.csv rec.Q_flow:2Q_field P_elec:2P_net``

The file names are separated by a colon (not space as variable names already soak up all extra arguments, and for now I didn't want to introduce a --vars flag for backwards compatibility).  An optional integer can be placed in front of a variable name to indicate which file the variable should be pulled from (might be common names).  The default is to pull from the first file.

Hourly data can be exported from SAM into a csv file easy enough, but that cannot be used directly at the moment.  First the time column should be converted to seconds from a date so that it matches the SolarTherm output, and the data labels should be converted to a single word and an optional units in brackets.  It would be easy enough to write a short script to do the conversion, but for now I'm just doing it in a spreadsheet and exporting it to a csv.

closed 2016-02-24 by paul-scott

paul-scott, 2016-02-26:
By the way, wrote the script to automate conversion of SAM csv data: `st_conv_sam_hourly`

It shortens and mangles the names so that they are manageable to write.

commit c0a5fd5b (2016-02-24): Fixed #42: plotting from csv