SolarTherm/SolarTherm #43: Investigate output power gap

paul-scott, 2016-02-24:
SAM outputs higher peaks in P_elec (c_ele_P_out_net) and less so rec.Q_flow (f_th_P_tot_prod).  Investigate why.

paul-scott, 2016-02-24:
Remember to set weather file start time to 1800 while comparing.

paul-scott, 2016-03-07:
The difference in P_elec was due to the use of the dry vs wet bulb temperatures in the power block deratings.  When SAM is changed to dry they match (start up consumption still not resolved).  At a later time we can build in features to select dry vs wet, but for now doing this is good enough for validation.

paul-scott, 2016-03-07:
In addition to making the polynomial representation of these deratings more general, I also used the total radiant energy incident on the receiver instead of the irradiance.  I assume $I_{bn}$ and $I_{bn, des}$ relate to the beam normal irradiance?  It seemed nonsensical to adjust the thermal performance according to this because it is not accounting for the optical performance of the field.  Maybe I'm wrong?

paul-scott, 2016-03-07:
Another differences is that the receiver thermal loss fraction is relative to the receiver incidence design value, and not the receiver output design value.  This is more natural, because it is 1 - eff at the design point.

paul-scott, 2016-03-07:
The biggest difference appears to originate back in the "Field thermal power incident (MWt)" value.  I would have expected this to be I_dni * A_field, but it is consistently 2.73% higher than this value (comparing SAM data to SAM data).

jdpipe, 2016-03-07:
Could this be something to do with the way the mirror reflectivity or spillage losses and being accounted?

paul-scott, 2016-03-07:
As far as I know these don't exist.  There are a "Cleanliness Factor" and a "General optical derate", but both of these are set to 1.  This extra amount of energy incident on the field shows up at the receiver + field output.

jdpipe, 2016-03-07:
Could this be due to the reference thermal loss fraction? I note that in my dish calc, it was equal to 0.02822 MWt/MWthcap. Interestingly 0.02822/(1+0.02822) = 0.02744, quite close to you 2.73% number...?

paul-scott, 2016-03-07:
This is for the trough model, which uses a value of 0.071, so I assume that is just a coincidence.  And it should be totally unrelated to the thermal loss.  SAM provides the field area it calculates and I can access the DNI it calculates, which I then use to compare to what it outputs for "Field thermal power incident".