SolarTherm/SolarTherm #45: Construct ASTRI reference system

paul-scott, 2016-03-15:


jdpipe, 2016-10-19:
We have now got 'reference_1.mo' and 'reference_1_fluid.mo'. We need to add test cases for these reference models in Travis to that we can check for regressions. The test cases should compare peak and average/annualised results from our model from the comparable values from SAM.

jdpipe, 2016-11-24:
Noting an issue with the current `reference_1.mo`: we don't get the required cost and output variables from the model when it is simulated with `st_simulate`. The result is that post-processing and optimisation tools from Paul won't yet work with this model. We should put in place checks for all our sample models to ensure they have the required output variables.