SolarTherm/SolarTherm #46: Consider methods for versioning models

paul-scott, 2016-03-15:


jdpipe, 2016-03-16:
- most users will have model files outside solartherm library, but they will like to know that the models they are importing from solartherm are the same version as they originally used.
- this could be implemented by embedding a file hash in the user's model file, which could be checked by some non-modelica code (find dependences, check for user-specified file hashes, warn in case of mismatch)
- this could also be implemented by a Modelica-aware continuous integration tool, which tracks dependencies between files and warns people when dependencies for their 'owned' files have changed in the repository.
Although this is a nice-to-have, I suspect that a few well-crafted test cases will be just as effective.

adelacalle, 2016-03-18:
What about "annotation" function? I tried it with good results in Dymola (with BondLib library), because the update of the library is automatic. I don't try with OMEdit.

jdpipe, 2016-03-18:
The problem I see is that changes to models can be made without a developer remembering to bump the model version/annotation. That seems to me to be the part that is hard to control.

thorade, 2018-04-12:
You might be interested in the ideas discussed here:
https://github.com/impact
http://dx.doi.org/10.3384/ecp14096543
http://dx.doi.org/10.3384/ecp15118725