SolarTherm/SolarTherm #48: A new folder-tree structure in Modelica library

adelacalle, 2016-03-17:
This was my proposal:
 
Sources (All solar sources, Weather…)
Collectors (Your optics)
Receivers
Storage
PowerBlocks
Economics (Models where we can calculate the costs)
Media
Fluid (All fluid components, i.e.: Pipes, pumps, valves, heat exchanger?…)
Control (State machines, PIDs…)
Utilities (No physical models: functions, mathematical or modeling utilities)
Examples (Runnable examples, maybe your general system can be place here)
UsersGuide (Documentation in html, Release notes, contact..)
Icons (where we can define special icons for the graphical interface)

As we talk by email, I’m agree “Solar” is implicit in the library, don’t have sense. Between “collectors” and “optics”, for me it’s more intuitive “collector” but I can accept “optics”. However, I’m sure that if we call HeliostatField to the component which models the field is better than SteeredConc or the other blocks.

I only talks about the Modelica library. I think the external folder-tree should be preserve it is better for scripts.

paul-scott, 2016-03-18:
UsersGuide -> UserGuide
Add a Resources folder

adelacalle, 2016-03-18:
I copy the structure from Fluid and they call this folder "UsersGuide". I don't know if it is correct English, sorry. 

Why do you need resources folder? Is it for the data? If yes, you don't need this folder because you can't read data in the editor browser.

paul-scott, 2016-03-18:
Ok sure stick with UsersGuide in that case.

Subfolders of Resources like Library and Include are the default locations for external C code.  It doesn't include a package.mo file so something like OMEdit ignores it.

Pretty much any example we want to provide will need to load an external data file.  The approach we have taken to date has been to keep the examples and data files out of the library.  When you want to run the examples you have to make sure you change the working directory to where the examples are so that it can find the data files using relative paths.  This is more in line with what users would be doing when they have their own data and create their own system models.

If we want to include them in the library, then Resources/Data is where they should go (have a look at the Modelica library in a file browser).  This way they can easily be referenced as "modelica://SolarTherm/Resources/Data/....", no matter where the current working directory is.

adelacalle, 2016-03-18:
Maybe I don't express well. I'm in line with you, I only talks about the folders included in package.mo.

paul-scott, 2016-03-21:
Just letting you know that I'm working on implementing these changes, also fixing up coding style and names for components.  Should be finished by the end of the day.

paul-scott, 2016-03-22:
These changes implemented in 42c28e05bd8bcad6e63fa86b3974d316073b61b1 (and later commits with model name adjustments).  One difference is that Analysis is used instead of Economics, as it contains Finance calculations and other energy performance calculations.

The Interfaces folder still exists (will go when WeatherSource uses inner/outer).  Also UsersGuide and Icons don't exist because we have nothing to put in them yet.

paul-scott, 2016-04-20:
I'm reopening because Alberto has some new package structure ideas.  Put these here so that we can discuss and implement them in the repository.