SolarTherm/SolarTherm #49: Modelica graphical interface

adelacalle, 2016-03-18:
With the addition of icons and diagrams (high-level modeling) we can provide users a friendly way to model their own systems. Same computational effort, more easy.

The problem, as Paul commented, is that if we are going to use and code editor (e.g. OMEdit) it inherently changes the files. I think the best way to handle with that is to save the model with the editor and upload the library. Once it is saved, next times the files wouldn't change (I hope). Can Paul check that?

paul-scott, 2016-03-18:
Yes that would make them all consistent, but consistent in a bad way in many occasions.

The nature of the modelica language often requires long lines, which need to be broken up into in order to be readable.  For example, take a look at the Modelica.Fluid.Pipes.mo file, both from OMEdit, and externally using a text editor.  OMEdit turns the flowModel instantiation into a single line making it unreadable.  When this is saved the original structure will now be destroyed.

I think this is a pretty big problem, what do you think?  I can't see anything in the OMEdit options to turn this off, but maybe you can take a look and see if you find anything.

adelacalle, 2016-03-18:
I understand what you said, but the problem is that you can't read correctly in github's website? Because the text editor and OMEdit can broke the lines allowing read all in the window. I'm sure that dymola preserves the original file, but we don't have license.

paul-scott, 2016-03-18:
Having them displayed nicely on github is a minor concern.  The main concern is that when we open them in any text editor (including the OMEdit editor) they will be difficult to read.  Maybe there is a hidden option that we are not aware of.  I'll ask on the OpenModelica forum.

paul-scott, 2016-03-18:
Understandably we are not the only ones with this problem:
https://www.openmodelica.org/forum/default-topic/1024-why-does-omedit-condense-code#p3826

Some tickets exist, but they are still not complete:
https://trac.openmodelica.org/OpenModelica/ticket/2439
https://trac.openmodelica.org/OpenModelica/ticket/2905

jdpipe, 2016-03-18:
Perhaps we can agree not to modify our model library in OMEdit, but only system models / flowsheets? Ideally, these flowsheet models won't have any local equations declared...?

By the way, what is the approach for adding up costs of items when a graphical model editor is used? Does Modelica provide summation operators, eg to iterate through all models refined from a particular parent class, for example?

jdpipe, 2018-01-26:
With the current version in alberto-merge branch, we are able to edit flowsheets from OMEdit and still run them through the st_simulate et al scripts. So this issue seems to be solved for now.

closed 2018-01-26 by jdpipe