SolarTherm/SolarTherm #51: Coding style guide

paul-scott, 2016-03-18:
Another thing we should consider doing is developing a coding style, in terms of how to name variables and the like.  So far I have been somewhat inconsistent, but so are the standard libraries.  For example, when to use camelCase vs underscore separation and the like.

adelacalle, 2016-03-18:
It's a very good observation. I can adapt myself to any rule. What it is the easiest rule to implement now? Maybe the camelCase.

paul-scott, 2016-03-18:
Some examples from the standard library:

Q_flow
p_a_start
p_b_start
m_flow_start
nParallel
pathLengths
crossAreas
dimensions
roughnesses
dheights

paul-scott, 2016-03-18:
My preference would be (which is similar to typical C++ conventions):

- Classes/models/packages: CamelCase
  - GenericSystem, SolarTherm
- Functions: camelCase (maybe, or maybe just _ separated all lower case)
  - checkVersion, getMetadata
- Variables/parameters:  _ separated with uppercase used if it is a standard physical symbol
  - Q_flow (heat flow), p_start (starting pressure), A_cross (cross sectional area), P_nom (nominal power)
- Constants: _ separated all upercase (or maybe just same as variables/parameters)
  - PI, CONSTANT_NUMBER

That said, I won't be working on the project much longer so let me know if you have issue with this.  Also John?

jdpipe, 2016-03-18:
My mild preference, for what it's worth, would be:
- ClassNames
- function_names and method_names (a bit pythonesque)
- OP, RC, ST1, ST2, PB -- names of major components in the system, short all-caps abbrevs, similar to Aspen/chemical process modelling conventions
- v_desc, something like the convention in Engineering Equation Solver -- a text approximation to mathematical nomenclature (my pref is to use 'epsilon_ext' for external surface emissivity, etc, but I note that Paul has used more descriptive things like 'emis' instead of 'epsilon' which is also fine)
- BIG_CONSTANTS_IF_NEEDED

On such matters though, the most important thing is consistency, and the second most important thing that that the person doing the coding most deserves to be the one who is comfortable with the convention.

paul-scott, 2016-03-22:
Added a coding style to the documentation.  For some reason it hasn't shown up on rtd with the last commit (will have to investigate tomorrow).  For now it can be found here: https://github.com/SolarTherm/SolarTherm/blob/master/doc/source/codestyle.rst

paul-scott, 2016-03-22:
Just pressed rebuild and it worked this time:

http://solartherm.readthedocs.org/en/latest/codestyle.html

closed 2016-03-22 by paul-scott