SolarTherm/SolarTherm #52: weatherbus and global variables

adelacalle, 2016-03-18:
I'm not sure that the actual weatherbus is a good practice to manage variables between submodels in the high-level modeling. Do you explore other solutions? For example, why do you not use inner and outer classes? In order to preserve the abstraction between models, some variables required by almost all the models (e.g. ambient temperature) can be a global variable of the system. Maybe this makes easier the balance of the models.

paul-scott, 2016-03-18:
I agree inner/outer is likely a cleaner approach in this case.

jdpipe, 2018-01-26:
Ali -- thought it might be good to review this issue given that we are working towards merging back to master shortly. Let's discuss.