SolarTherm/SolarTherm #55: Solar position calcs

paul-scott, 2016-04-06:
Following copied from email.

Alberto

I just looking the code and I found that the equations of the solar position with the Duffie & Beckman algorithm are not the same than any of the provided references:
 
              // Solar Engineering of Thermal Proceesses - Duffie and Beckman
               // http://www.esrl.noaa.gov/gmd/grad/solcalc/solareqns.PDF
               // http://ascend4.org/Calculation_of_sun_position
               // http://www.pveducation.org/pvcdrom/properties-of-sunlight/suns-position
 
What reference do you use for this algorithm?

paul-scott, 2016-04-06:
John:

I looked quickly at the code you mentioned,
https://github.com/SolarTherm/SolarTherm/blob/master/SolarTherm/Sources/Solar.mo

The source is Duffie and Beckman, as noted in the first line you sent.

Line 49 (E) of the file above matches Eq (1.5.3) in my edition (3rd Ed).
Line 53 (dec) matches  Eq (1.6.1b)
Line 73 (zen) matches Eq (1.6.5)
Line 75 (azi) matches Eq (1.6.6)

I'm not clear on the ecliptic longitude, perhaps that is mentioned on one of the webpages.

Although Duffie and Beckman's approach is simplified and perhaps not as accurate as others like the NREL code, I am quite certain that these equations will be accurate enough for annual performance estimation, which is our goal here.

paul-scott, 2016-04-06:
Alberto:

The B equation (1.4.2) it’s not the same equation.
 
Yours is:
B = 2*pi*(to_hour(t) - tzone - 12)/(24*365);
 
In the book is:
B=(n-1)*360/365 (in degrees)
Where n it’s the day
 
In radians should be:
B=2*pi*(to_day(st)-1)/365
Or
B=2*pi*(to_hour(st)-24)/(24*365)
 
Where st is the standard time (yours to_hour(t)-tzone?). Maybe the 12 it’s a mistake or my calculation is wrong.
 
The next question is why do you calculate the solar time like this?
t_solar = to_hour(t) + lon/ang_vel - tzone + E;
 
in the book is:
Solar time − standard time = 4(Lst− Lloc) + E
 
where Lst is the standard meridian for the local time zone, Lloc is the longitude of the
location in question, and longitudes are in degrees west, that is, 0◦< L < 360◦. The parameter E is the equation of time (in minutes).
 
If think the problem it’s the management of the Modelica time because is confused. Now I don’t know if it’s the local time, standard time(without winter and summer changes), the GMT time or what?

paul-scott, 2016-04-06:
The D&B one uses discrete days instead of a continues time in the B calc.  Looking at other references I think I decided that the zero point should be solar noon on the first day.  Half a day should make little difference to the end result, so at the end of the day it probably doesn't really matter.  Take a look and let me know if you think that makes sense.

The second issue I don't know off the top of my head.  I'll go back and have a look (possibly not till next week).

The last thing I will say is that the unit tests should pick up obvious problems with the calcs.  If you do decide to change something, just make sure it still passes this test.

adelacalle, 2016-04-08:
Maybe there is a mistake with the ecliptic longitude. You can calculate it with the axial tilt and the declination (https://en.wikipedia.org/wiki/Position_of_the_Sun). The mistake is that your declination angle is in degrees and Modelica's sin function works with radians. Take a look of this. Probably this is the origin of the problem solved with the min/max functions.

paul-scott, 2016-04-08:
Why do you think the Dec is in degrees? I don't have all the details in front of me, but checking the code I'm pretty sure it is in radians.  Note that it might differ from what is in the book if I'm using a radians version of the calc.

The min/max's are to prevent numerical errors from causing problems when a very small but negative argument is produced.  These only occur in some instances and it will depend on the solver.

I haven't tested the ecliptic longitude calcs as much as the others so it is possible there is another problem.  This was created for the interface with Clothilde's stuff, but we might not end up using ecliptic longitude.

adelacalle, 2016-04-08:
Sorry, my mistake. Maybe it's confused to have the half of the angles in degrees and the other half in radians.

paul-scott, 2016-04-08:
If you find a better way to do/explain/comment things you should start making changes and submitting commits as you go. Otherwise we will forget and someone else will have the same problem.

closed 2016-04-20 by paul-scott