SolarTherm/SolarTherm #59: st_simulate fails to abort

jdpipe, 2016-09-01:
Nicole is finding that when running a simulation (her own file in 'examples' directory), she gets an error from Weather.mo complaining about a version mismatch for the weather file (she has a .motab file that she has hand-crafted herself from an Excel input file -- perhaps an unexpected use-case).

What happens after the .motab file error though is the real problem: the st_simulate file keeps going, because it calls a .makefile file and from there seems to run the full simulation. If Nicole is running st_simulate in an uncleaned directory, then an old .makefile file can be there, and it gets run, and she can't be assured that she's getting proper valid current results any more.

Need to fix st_simulate or related files to ensure that the errors triggered from Modelica properly halt the simulation process.

paul-scott, 2016-09-01:
For the first part the weather file reader does some checks on the headers in the .motab file.  If it doesn't match what it expects then it throws an error.  This was designed as a safety check in case new weather file .motab structures/versions are adopted in the future.

Yep I've had this happen to me also and it can be misleading if you don't know that it can happen.  As you say we should propagate the errors to the top level.

jdpipe, 2016-10-19:
fixed, changed 'call' to 'check_call'.

closed 2016-10-19 by jdpipe