SolarTherm/SolarTherm #60: weather files being copied around between different 'resources' folders

jdpipe, 2016-09-01:
Nicole has been getting frustrated (understatement) by some puzzling behaviour by cmake and/or make. She has a weather file in examples/resources/weather which keeps getting overwritten every time she runs cmake -D[...] && make install. Why is this happening? The process of 'making' solartherm arguably shouldn't involve copying copies of files around inside the source tree, should it? She is not using a separate 'build' tree, which I believe you (Paul) preferred, so that is down to me, but do you see a way that we could fix this to make it more intuitive?

Which folder is the 'definitive reference' for .motab files, and why do we need multiple copies?

paul-scott, 2016-09-01:
Yes so I haven't ever tested building in the source directory, I always used a build directory:

mkdir build
cd build
cmake .. -D[...]
make install

I set up cmake to copy the resources folder into varies locations under the build directory, but I guess if that also happens to be your source directory it does unintuitive things.  If you look at ./resources/CMakeLists.txt there are 3 copies:

file(COPY ./ DESTINATION .)
file(COPY ./ DESTINATION ${CMAKE_BINARY_DIR}/tests/resources/)
file(COPY ./ DESTINATION ${CMAKE_BINARY_DIR}/examples/resources/)

The example and test models look for resources at a relative path to where they are executed (./resources/....).  Basically I put these copies in so that I could run the examples and tests from either: /build/, /build/examples/ or /build/tests/.  This was for convenience but a more robust solution is warranted, perhaps setting up an environment variable or something to point to the resources path.

The original directory that gets copied is /resources.  I imagine that if you are building in the source directory and put your new files in here and run cmake each time you make a change then everything will be ok (will get copied to /examples/resources/ and /tests/resources/).

Alternatively just set up a new folder for doing your experiments and manually put your model and resources into it (perhaps starting with a copy of the resources folder).  As long as you pull in the environment variables you can work from wherever you want.  This is how I think we would expect a typical user would set up and work on a new project (unless of course they are developing it for inclusion into the SolarTherm).

jdpipe, 2016-09-15:
I think that an environment variable sounds like a good approach, possibly combined with some function, defined somewhere, that helps to establish a reasonable default location for the 'resources' in the absence of the env var being set.