SolarTherm/SolarTherm #66: travis configuration broken for 'trusty' distro

jdpipe, 2018-02-28:
The old Ubuntu 12.04 'precise' distro is no longer supported by Ubuntu, and support is being discontinued by Travis as well. We need to move to their 14.04 distro instead.

The problem is that we see errors from Travis when we update to trusty, as shown below.

This appears to be related to Travis bug https://github.com/travis-ci/travis-ci/issues/8507.

```
12.09s$ sudo apt-get update -qq
W: The repository 'http://ppa.launchpad.net/couchdb/stable/ubuntu trusty Release' does not have a Release file.
W: Failed to fetch http://ppa.launchpad.net/chris-lea/redis-server/ubuntu/dists/trusty/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch https://packagecloud.io/computology/apt-backport/ubuntu/dists/trusty/InRelease  Failed to connect to packagecloud.io port 443: Connection timed out
W: Failed to fetch http://ppa.launchpad.net/git-core/ppa/ubuntu/dists/trusty/InRelease  Could not connect to ppa.launchpad.net:80 (91.189.95.83), connection timed out
W: Failed to fetch http://toolbelt.heroku.com/ubuntu/./InRelease  Unexpected type excepted 21 != 5
W: Failed to fetch http://ppa.launchpad.net/openjdk-r/ppa/ubuntu/dists/trusty/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://build.openmodelica.org/apt/dists/trusty/InRelease  Could not connect to build.openmodelica.org:80 (130.236.180.159), connection timed out
W: Failed to fetch http://ppa.launchpad.net/pollinate/ppa/ubuntu/dists/trusty/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/webupd8team/java/ubuntu/dists/trusty/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/couchdb/stable/ubuntu/dists/trusty/main/binary-amd64/Packages  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/couchdb/stable/ubuntu/dists/trusty/main/binary-i386/Packages  Unable to connect to ppa.launchpad.net:http:
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

jdpipe, 2018-03-07:
For now, this networking error has gone away, not sure if it's fixed or not though.

thorade, 2018-04-12:
Also, OpenModelica does not offer recent releases for Ubuntu 12.04:
https://openmodelica.org/forum/default-topic/2066-installation-on-ubuntu-12-04-or-14-04

A solution could be to use docker, then you could switch to any Ubuntu LTS release like 16.04 or in a few weeks even 18.04 BionicBeaver. Those docker images could also be used for local testing, because you have the identical setup.

thorade, 2018-04-12:
Another idea: Does it make any difference whether you use sudo or the Travis APT Addon?
https://docs.travis-ci.com/user/installing-dependencies/#Installing-Packages-with-the-APT-Addon

Here is an example:
https://github.com/thorade/ExtraMath/blob/travis_OM_generic/.travis.yml

ashirazi7, 2018-07-08:
Resolved!

closed 2018-07-08 by ashirazi7