SolarTherm/SolarTherm #73: Unable to open examples of the Running SolarTherm Guide

sun-ai, 2019-12-10:
Hi Everyone,

I was trying to open the example in "Running SolarTherm Guide" (https://github.com/SolarTherm/SolarTherm/wiki/Running-SolarTherm-via-OMEdit) via OMEdit, but I am getting an error (in attachment the scripting error). 

I am running latest Ubuntu (18.04.3-desktop-amd64) in a Virtual machine.
[image]
[log.txt](https://github.com/SolarTherm/SolarTherm/files/3946772/log.txt)

Any idea of the issue? Thanks for the help!

jdpipe, 2019-12-11:
Hi Sun
Thanks for the report -- I've passed this issue on to Armando, and he's making some changes that will fix this issue. Perhaps it's arisen due to changes in recent OpenModelica versions.
Cheers
JP

sun-ai, 2019-12-11:
Hi JP,

Thanks for your flash response! Nice to hear that, i will wait for the update :)

Nice week!