SolarTherm/SolarTherm #78: st_optimise: NameError: name 'scale' is not defined

arfontalvoANU, 2020-11-03:
After running the following command on the Linux terminal:

`st_optimise Reference_2.mo --excel /home/arfontalvo/program_files/multiprocessing/examples/Reference_2_params.xlsx --wd /home/arfontalvo/Reference_2_opt  --objective lcoe,+capf --method dakota_moga`

It shows the following error:

```
Traceback (most recent call last):
  File "/home/arfontalvo/.local/bin/st_optimise", line 268, in <module>
    args.v, scale, offset, perf_i, par_n, resultclass, args.method, obj_n, obj_sign, args.test)
NameError: name 'scale' is not defined
```

Machine: `Dell Optiplex 7050`
System: `Ubuntu 16.04`
Python version: `2.7`
Branch: `st_mp`

yewang726, 2020-11-16:
Hi Armando, 

I think I've just solved this error. Could you please check the latest st_mp branch? It will be merged to the master branch soon.

Cheers,
Ye

arfontalvoANU, 2020-11-17:
Hi Ye

I'm using the lattest version of st_mp:

`https://github.com/SolarTherm/SolarTherm/commit/420b6099202d5d48258ed5ba41893fb7f414aa6d`

And the issue `NameError: name 'scale' is not defined` is already fixed.

Thank you very much!

Cheers,

Armando

closed 2020-11-17 by arfontalvoANU