SolarTherm/SolarTherm #79: SimpleSystem not working with OMC 1.17.0

arfontalvoANU, 2021-03-31:
**Abdul Azidin got the following error when running the SimpleSystem.mo:**

```
Compiling model
Notification: Automatically loaded package Complex 3.2.2 due to uses annotation.
Notification: Automatically loaded package ModelicaServices 3.2.2 due to uses annotation.

Error processing file: /home/azimazidin/solartherm/examples/SimpleSystem.mo
[/home/azimazidin/solartherm/examples/SimpleSystem.mo:226:3-229:11:writable] Error: A when-statement may not be used inside a function or a while, if, or for-clause.

# Error encountered! Exiting...
# Please check the error message and the flags.

Execution failed!
Traceback (most recent call last):
  File "/home/azimazidin/.local/bin/st_simulate", line 167, in <module>
    sim.compile_model(args=mcargs)
  File "/home/azimazidin/.local/lib/python3.8/site-packages/solartherm/simulation.py", line 274, in compile_model
    sp.check_call(['omc', '-s', '-q', '-n='+str(n_proc)]
  File "/home/azimazidin/anaconda3/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['omc', '-s', '-q', '-n=0', '-i=SimpleSystem', '/home/azimazidin/solartherm/examples/SimpleSystem.mo', 'Modelica', 'SolarTherm']' returned non-zero exit status 1.
```

Operation system: Pop!_OS 20.04 LTS
OMC version: 1.17.0

arfontalvoANU, 2021-03-31:
We managed to run the model SimpleSystem.mo by removing OMC 1.17.0 and installing OMC 1.16.0.

jdpipe, 2021-08-23:
I'm currently running all the SolarTherm tests with OpenModelica 1.18.0~dev.beta1-3-g75b5528 on Ubuntu 20.04, and they are working for me.

closed 2021-09-02 by arfontalvoANU

jdpipe, 2021-09-02:
Thanks Armando... we must learn how to invoke omc via the new frontend, I guess.