SolarTherm/SolarTherm #89: memory leak in on-the-fly surrogate

jdpipe, 2021-11-02:
Hey Philipe -- wanted to flag this and ask which line is correct... 
https://github.com/SolarTherm/SolarTherm/blob/28bb2fefffa2f83d5b54c318aea9b651999f7dda/src/modelica/st_on_the_fly_surrogate.c#L2919

philgun, 2021-11-02:
Hi John,

I think the implementation is correct. OD_array is ssc_number_t type to contain all off-design sCO2 PB input to be simulated shall the system generate a new surrogate model.

Line 2918 is to allocate memory to host OD_array.. The value of OD_array is then instantiated to be the value of sim-> array_OD. The value of sim->array_OD is createdin line 2914 (calling function genOffDesignArray).

jdpipe, 2021-11-03:
Have a look at this:
```
		OD_array = NEW_ARRAY(ssc_number_t, sim->rows_OD*6);
		OD_array = sim->array_OD;
```
This code is allocating some memory to `OD_array`, but is then allocating another pointer, `sim->array_OD` to that same variable. That means that you lose the pointer to the allocated memory from the first call. It is a lost pointer; you don't know what it was pointing to, so you can't free it even if you want to.

philgun, 2021-11-03:
I see. My intention is to create a sim_struct and within this sim_struct, there is a member called array_OD. A function called genOffDesignArray populates the array_OD within the sim_struct. Then I want to transfer the value of sim_struct array_OD to OD_array.

OD_array is then used to set the off-design case 
```ssc_data_set_matrix( data, "od_cases", OD_array, sim->rows_OD, 6); /*user Off design cases*/```

Should I skip the OD_array instantiation and use this instead:
```ssc_data_set_matrix( data, "od_cases", **sim->array_OD**, sim->rows_OD, 6); /*user Off design cases*/```

jdpipe, 2021-11-04:
The issue here was that you were allocating memory, but then you weren't correctly assigning the pointer to that memory in a way that made it accessible later. I changed it to what I *think* you meant. Hopefully it's OK for now.

closed 2021-11-04 by jdpipe