SolarTherm/SolarTherm #93: Excecution failed solartherm examples. Some library version issue

FNietoUC, 2022-01-09:
Hi everyone,
I've tried to run some example following the steps of the link:
https://github.com/SolarTherm/SolarTherm/wiki/Running-SolarTherm-on-Windows-%28MSYS2%29

Apparently, I have done all the steps correctly but when I want to test some simulation, I just got an error.
Here is my output. I realized that I have the 10.2.0 instead of 5.3.0 in /lib/gcc directory and the problem between the 3.2.2 and 3.2.3 loaded libraries.
```
# st_simulate Reference_1.mo
Compiling model
We could not find some needed MINGW paths in $OPENMODELICAHOME or $OMDEV. Searched for paths:
        C:/Program Files/OpenModelica1.18.0-64bit/\tools\msys\mingw64\bin [found]
        C:/Program Files/OpenModelica1.18.0-64bit/\tools\msys\mingw64\lib\gcc\x86_64-w64-mingw32\5.3.0 [not found]
Notification: Automatically loaded package Complex 3.2.2 due to uses annotation.
Notification: Automatically loaded package ModelicaServices 3.2.2 due to uses annotation.
Warning: Requested package Modelica of version 3.2.3, but this package was already loaded with version 3.2.2. You might experience problems if these versions are incompatible.

Error processing file: C:/msys64/home/Mauricio/solartherm/examples/Reference_1.mo
Warning: Requested package Modelica of version 3.2.3, but this package was already loaded with version 3.2.2. You might experience problems if these versions are incompatible.
[C:/msys64/home/Mauricio/.local/lib/omlibrary/Modelica 3.2.2/Media/package.mo:4521:11-4524:41:writable] Error: Type mismatch for positional argument 3 in assert(level=AssertionLevel.error). The argument has type:
  enumeration(warning, error)
expected type:
  enumeration(error, warning)
Error: Error occurred while flattening model Reference_1

# Error encountered! Exiting...
# Please check the error message and the flags.

Execution failed!
Traceback (most recent call last):
  File "C:\msys64\home\Mauricio\.local\bin\st_simulate", line 175, in <module>
    sim.compile_model()
  File "C:/msys64/home/Mauricio/.local/lib/python3.9/site-packages/solartherm/simulation.py", line 293, in compile_model
    sp_run(call)
  File "C:/msys64/mingw64/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['omc', '-s', '-q', '-n=0', '-d=nonewInst', '-i=Reference_1', 'C:/msys64/home/Mauricio/solartherm/examples/Reference_1.mo', 'Modelica', 'SolarTherm']' returned non-zero exit status 1.
```
In any case, I've tried to run directly from the OMEdit but I get this error output

```
[1] 10:26:27 Scripting Notification
Automatically loaded package Modelica 3.2.3 due to uses annotation.

[2] 10:26:27 Scripting Notification
Automatically loaded package Complex 3.2.3 due to uses annotation.

[3] 10:26:27 Scripting Notification
Automatically loaded package ModelicaServices 3.2.3 due to uses annotation.

[4] 11:16:21 Scripting Error
Failed to insert class SimpleSystem within examples;  the available classes were:
  SolarTherm.Systems.GenericSystem
etc
etc
...
...
```
If you can find any solution, please let me know

Best regards,

jdpipe, 2022-01-11:
Hi F,

We're in the process currently of working with the OpenModelica people to get an updated version of the binaries for MSYS2 built. In the meantime, we've seen some difficulties with running on Windows with our current instructions. You might have a bit more luck, possibly, if you use the 'particle-master-intermediate' branch of our SolarTherm code, though -- it would be worth a try. 

Can I confirm that SimpleSystem.mo does work for you on your system, though?

Cheers
JP

jdpipe, 2022-01-11:
Possible duplicate of #92.

FNietoUC, 2022-01-13:
jdpipe I realized that I can have access to examples if I open the package file located in:
C:\msys64\home\User\solartherm\examples

After that I tried to open the SimpleSystem.mo example in OMEdit and a message box was opened with the next: 

_

> The code generation process failed, 
> Most likely this is due to some issue in the Modelica source code .... or in OpenModelica compiler frontend.
> Options: Continue with current version; Try with older frontend; Switch permanently to older

_ 
I chose: Try with older, and It starts to simulate.

It last around 2 min to solve it with no errors. I'm able to plot all the variables correctly

I tried to do the same with Reference_1.mo but I have errors like this
Warning: maximal number of iteration reached but no root found
However, I can plot the results too. The same in Ubuntu 18.04