SolarTherm/SolarTherm #96: Some issues with Solartherm installation in Ubuntu 18.04 and run examples

Ijarias, 2022-02-01:
Hi everyone, 
I have  been trying to work in SolarTherm in ubuntu 18.04. Therefore, I have followed this installation instruction 
https://github.com/SolarTherm/SolarTherm/blob/master/doc/source/installation.rst

Although I have followed the instructions to the letter. I have had some problems with the dakota installation. When I put this line command **dakota --version,** my terminal returns  the following message - **dakota: error while loading shared libraries: libgsl.so.23: cannot open shared object file: No such file or directory.**
[image]

Furthermore, I don't fully understand the last instructions of the point **5. Build and install SolarTherm**. According to what I understand, I must put the following commands: 

**scons PREFIX=~/.local**
**scons install PREFIX=~/.local**

However, in the following two lines I don't understand what I should insert in the highlighted part in bold.  

scons PREFIX=~/.openmodelica/libraries INSTALL_OMLIBRARY=**/the/directory/where/Openmodelica/installed**
scons install PREFIX=~/.openmodelica/libraries INSTALL_OMLIBRARY=**/the/directory/where/Openmodelica/installed**

In a first instance, insert the directory where the Omedit files are installed. which in my case is **/usr/lib/omlibrary/Modelica 3.2.3**. However, I obtained this message:

[image]

Finally, as solartherm was correctly clone in my laptop, I have tried to run the examples using OMEdit. Therefore, I followed this instructions:

https://github.com/SolarTherm/SolarTherm/wiki/Running-SolarTherm-via-OMEdit

Nevertheless, when  the SaltSCO2System.mo example is runnig I obtained this message in Omedit: 
 
**[examples.SaltSCO2System: 197:27-197:40]: Modified element unit not found in class Integer.**

so I don't understand if this is a fatal error or what causes it.

I hope you can help me solve these problems.
kind regards.

jdpipe, 2022-02-01:
Hi Ignacio,
Regarding the first error, `dakota --version` is giving an error because it is linked to `libgsl.so.23`, but that library cannot be found at runtime. Perhaps you need to `sudo apt install libgsl23` or `sudo apt install libgsl`? FYI, here is what I have on my local (20.04) system:
```
john@ovo:~$ find /usr/lib -name "libgsl*"
/usr/lib/x86_64-linux-gnu/libgsl.so.23
/usr/lib/x86_64-linux-gnu/libgsl.so
/usr/lib/x86_64-linux-gnu/libgsl.a
/usr/lib/x86_64-linux-gnu/libgslcblas.a
/usr/lib/x86_64-linux-gnu/libgsl.so.23.1.0
/usr/lib/x86_64-linux-gnu/libgslcblas.so.0
/usr/lib/x86_64-linux-gnu/libgslcblas.so.0.0.0
/usr/lib/x86_64-linux-gnu/libgslcblas.so
```
Regarding the errors with the salt-sCO2 system, I've CCed Armando to see if he can help.
Cheers
JP

arfontalvoANU, 2022-02-02:
Hi @jdpipe and @Ijarias 

To avoid the following error:

**[examples.SaltSCO2System: 197:27-197:40]: Modified element unit not found in class Integer.**

we should allow the user needs to enable the old "FrontEnd" when running `st_simulate`. @yewang726 included this option before, but somehow it has been removed.

For this specific system, I can modify the model to run with the new FrontEnd. However, @Ijarias could get this error again when running models like the `SimpleSystem`.

Ijarias, 2022-02-02:
Hi @arfontalvoANU  and @jdpipe 
Thank you for your answers. Thanks to these I already solved dakota's problem. However, I would like to understand the issue related to these lines of installation code. 

scons PREFIX=/.openmodelica/libraries INSTALL_OMLIBRARY=/the/directory/where/Openmodelica/installed
scons install PREFIX=/.openmodelica/libraries INSTALL_OMLIBRARY=/the/directory/where/Openmodelica/installed

I tend to think that in the part INSTALL_OMLIBRARY = /the/directory/where/Openmodelica/installed, the Omedit libraries's path = /usr/lib/omlibrary/Modelica 3.2.3 is not the correct, or else the way to insert it is not correct.

On the other hand,  Currently I'm a PhD student in the Pontifical Catholic University of Chile (PUC) and, I'm working together a the professors Dr. Eduardo Zarza and Dra. Loreto Valenzuela of the Plataforma Solar de Almeria and Dr. Jose Cardemil and  Dr. Rodrigo Escobar of PUC.  Mentioned the above, I would like to know if it is possible to have a brief meeting of no more than 30 minutes with @arfontalvoANU via Zoom, when it suits you best. In order to ask you some questions concerning the program, among other things. We can speak in both English and Spanish.

I leave you my contact number to talk by whatsapp or telegram and my email to be able to coordinate.

Cell phone= +56955134350

email= ivarias@uc.cl

jdpipe, 2022-02-03:
Hi Ignacio

Your strategy of setting PREFIX using SCons is not quite right. SCons is installing scripts and a Python module and other things, not just Modelica files, so you should be using

scons PREFIX=$HOME/.local

if anything (which is the default), to ensure that those files go to a good place. A good choice of PREFIX is a path such that $PREFIX/bin is already in your  environment's PATH, which means that our convenience script 'st' will immediately be executable from the command line after installation is completed. You can editor your ~/.profile or ~/.bashrc to set this PATH if needed. The idea here is that as long as 'st' is in your path, you shouldn't have to set anything else in your environment.

You can use INSTALL_OMLIBRARY if you want the Modelica files to be installed somewhere non-standard (our default is ~/.openmodelica/libraries, which is a directory that the commandline 'omc' compiler looks in by default). Or, you can leave that set to the default, and tell OMEdit where those 'installed' files are located from. Another option would be not install SolarTherm at all, and simple to access the associated model files from directly in the source tree.

I know that @arfontalvoANU uses OMEdit more than I do; perhaps he or @yewang726 can give more advice on a good workflow for accessing SolarTherm from OMEdit. I normally just use our 'st simulate' script, for which OMEdit is not used, and the OM compiler, 'omc', is used directly. Getting OMEdit to work correctly may possibly be a little bit tricky in cases where external libraries are needed, additional PATH and PYTHONPATH settings may be needed in your environment variables.

I note that we have some suggestions on running SolarTherm from OMEdit shown here. Perhaps it needs updating though (@yewang726 ?)
https://github.com/SolarTherm/SolarTherm/wiki/Running-SolarTherm-via-OMEdit