SolarTherm/SolarTherm #97: Help with st_plotmat function

Ijarias, 2022-02-16:
Hi everyone!, 
Currently I've been working  with solartherm. Then,  to internalize myself in the software  I have replicated the SimpleSystem.mo model following the tutorial indicated in the following link: https://github.com/SolarTherm/SolarTherm/blob/master/doc/source/tutorial.rst. 
Nevertheless, when I try to run the following line code "**st_plotmat SimpleSystem_res_0.mat wea.wbus.dni:P_elec,Q_flow_chg E**" the console gives me the following message:
[image]

To fix this I have tried the following:

1. I accessed to the directory "**/home/ignacio/SolarTherm/src/python/solartherm**" and opened plotting.py and uncommented the sections:

   matplotlib.rcParams['text.usetex'] = True
   matplotlib.rcParams['text.latex.unicode'] = False
   matplotlib.rcParams['font.family'] = 'serif'
   matplotlib.rcParams['font.serif'] = 'Times New Roman'
   matplotlib.rcParams['font.serif'] = 'cmr10'

2.  After the above, I save the changes and run the following line codes for the changes to take effect at Solartherm: 
    cd SolarTherm
    scons
    scons install

Nevertheless, I obtained the same error. I know that the problem is in matplotlib, but it would be nice to have your feedback to solve this. In the same way I have noticed that there is more than one plotting.py file, maybe I edited the one in the wrong directory. 

I will be attentive to your comments.
Kind regards.

jdpipe, 2022-02-16:
Hi Ignacio

It looks like we need to remove that 'text.latex.unicode' command, as noted [here](https://stackoverflow.com/questions/53157355/what-replaces-text-latex-unicode).

I wonder if you might also have some recently-deprecated settings in your ~/.matplotlibrc file?

Cheers
JP

Ijarias, 2022-02-16:
Hi John
I hope that you and your loved ones are in good health. That said, thank you very much for your response. You are right ! I have solved it.
Kind regards.
I.Arias