SolarTherm/SolarTherm #102: Modified element unit not found in class Integer.

jljines, 2023-04-21:
Hi,

Please, I was uploading my scripts so I have a issue : "[Modified element unit not found in class Integer.]", what does it mean?

thanks for your help

philgun, 2023-07-17:
Hi, my name is Phil and I am one of the developers of SolarTherm, in particular, the hybrid-pv-csp and particle-master-intermediate branch. Your problem rises up because you probably use the Modelica Standard Library (MSL) > 3.2.2. If you revert back to <= 3.2.2., this error should disappear. It happened to me once and this is my solution.

JoccabLison, 2025-01-06:
hi Phil，I wanna konw if there is a version based on MSL4.0.0,cause I also have other models using MSL4.0.0,so if there is a solartherm base on 4.0.0?

philgun, 2025-01-06:
Hey mate thanks for reaching out.

Unfortunately, no. It's all based on Modelica 3. I reckon the error comes
from one of the custom variables in the system level code e.g. horizon
(int) or t_storage (int).

If you get rid of tthose customised variables and change it to parameter
Real t_storage = 16.0 for example, all should work.

Cheers,
PG

JoccabLison, 2025-01-06:
Hello I have another problem, When I simulating SaltSCO2System in Example package, found an erro,Use of undeclared variable: eff_cyc In class examples.SaltSCO2System.
File: D:/My Sources/Dymola-working-directory/SolarTherm-master/SolarTherm-master/examples/SaltSCO2System.mo, line 2.
I check the Text, found the variable eff-cyc has no definition.How to solve it?The eff_cyc is a matrix or a algo or an equation？

philgun, 2025-01-06:
Hey,

I believe eff_cycle refers to power block cycle thermal efficiency (before
parasitic such as power electronics energy consumption).

I reckon eff_cyc has to be defined elsewhere.

Copying my other solartherm squad for better visibility.

Cheers,
PG