SolarTherm/SolarTherm #103: Run external fuction

Ijarias, 2023-06-22:
Hi every one, 
I'm trying to run in openmodelica an external python function through  a .c file, I  hope use this fuction together solartherm models. So I started with a simple py function to test this kind of integration, I attached the python and .c at the bottom.

I'm compiling my .c file in ,so through the following commands:
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c /home/srspoke/Desktop/Example_Compilance_C/Function2.c -o build/temp.linux-x86_64-3.6/home/srspoke/Desktop/Example_Compilance_C/Function2.o
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.6/home/srspoke/Desktop/Example_Compilance_C/Function2.o -o /home/srspoke/Desktop/Example_Compilance_C/libFunction2.so

Nevertheles, I recieve the following error message to run it in openmodelica:

/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyDict_SetItemString'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyModule_GetDict'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyUnicode_FromString'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyErr_Print'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyObject_CallObject'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyCallable_Check'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyDict_New'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyImport_Import'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `Py_Finalize'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `Py_Initialize'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyFloat_FromDouble'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyDict_GetItemString'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyFloat_AsDouble'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyTuple_SetItem'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyTuple_New'
/home/srspoke/Desktop/Example_Compilance_C/libFunction2.so: undefined reference to `PyErr_Occurred'

According with what I know, this occurs due to my .c program is not linked with the python header libs so i tried  to repair it through the following command,  which generate a .o and from which I build the .so:

gcc -c -L/usr/lib/python3.6/config-3.6m-x86_64-linux-gnu/libpython3.6.so -I/usr/include/python3.6m -lpython -fPIC Function2.c -o Function2.o

gcc Function2.o -shared -o libFunction2.so

I can not see what I'm doing wrong.  I know that your group have a great experience doing this kind of integration, so I would like to know what kind of commands usually you use to build the .so file ?. Becouse I tend to think that my problem is in the commands.

I will be grateful for any feedback you can give me.
kind regards.

**my python function looks like:**

def set_param(inputs={}):
    class Parameters:
        def __init__(self):
            self.X = None
            self.Y = None

    pm = Parameters()

    for k, v in inputs.items():
        if hasattr(pm, k):
            setattr(pm, k, v)
        else:
            raise RuntimeError("invalid parameter '%s'" % k)

    return pm

def mult(inputs={}):
   pm=set_param(inputs)

   print('')
   print('Test inputs')
   for k, v in inputs.items():
       print(k, '=', getattr(pm, k))
   print('')
   print('')
     
   return pm.X*pm.Y

if __name__ == '__main__':
    inputs = {'X': 5, 'Y': 10}
    
    l = mult(inputs)

**my .c file looks like:**

#include <Python.h>
#include <stdio.h>

double MyFunction(const char* pname, const char* pfunc	
		      , int argc
		      , const char* varnames[]
                      , const double var[]
){
    PyObject *pName, *pModule, *pDict, *pFunc, *pArgs, *pValue, *pResult;

    // Inicializar el interprete de Python
    Py_Initialize();

    // Cargar el módulo Python desde el archivo .py
    pName = PyUnicode_FromString(pname);
    pModule = PyImport_Import(pName);
    Py_DECREF(pName);

    if (pModule != NULL){
        // Obtener el diccionario global del módulo cargado
        pDict = PyModule_GetDict(pModule);

        // Obtener la referencia al objeto de función especificado por pfunc
        pFunc = PyDict_GetItemString(pDict, pfunc);

        if (pFunc && PyCallable_Check(pFunc)){
            // Crear una tupla vacía para los argumentos
            pArgs = PyTuple_New(argc);

            // Crear un diccionario vacío para los argumentos
            pValue = PyDict_New();

            for (int i = 0; i < argc; i++){
                // Agregar los valores de entrada al diccionario
                PyDict_SetItemString(pValue, varnames[i], PyFloat_FromDouble(var[i]));
            }

            // Establecer la tupla de argumentos en la posición 0
            PyTuple_SetItem(pArgs, 0, pValue);

            // Llamar a la función especificada por pfunc con los argumentos
            pResult = PyObject_CallObject(pFunc, pArgs);

            // Obtener el valor de salida de la multiplicación
            double result = PyFloat_AsDouble(pResult);
            return result;
            // Liberar memoria
            Py_DECREF(pArgs);
            Py_DECREF(pResult);
        }
        else{
            if (PyErr_Occurred())
                PyErr_Print();
            fprintf(stderr, "No se pudo encontrar la función '%s'\n", pfunc);
        }
        // Liberar memoria
        Py_DECREF(pModule);
    }
    else{
        PyErr_Print();
        fprintf(stderr, "No se pudo cargar el módulo '%s'\n", pname);
    }

    // Finalizar el interprete de Python
    
    Py_Finalize();
    
}

jdpipe, 2023-09-07:
Hi Ignacio!
So sorry this bug report has not attracted any notice until now. I would suggest you look at the `run_py_func.c` file in the folder `src/modelica` of the current master branch. This shows how we compile and link the script that, for example, runs a Python script that handles our ray-tracing. We compile this file using [SCons](https://scons.org/), so take a look at the `SConscript` file in that same folder.
Having said all this, I note that we are currently seeing a bug with that script, only on MSYS2. I should write that bug up so you can track it if you need to.
Cheers
JP

jdpipe, 2023-09-07:
Also note that we use the command `pkg-config python-3.8-embed --libs --cflags` to tell us what GCC flags are needed to link against Python. See the main `SConstruct` file for that stuff (~line 425)