SolarTherm/SolarTherm #113: The link provided for the installation of solstice and dakota is broken

xianping1993, 2024-03-10:
Hi!

I have been trying to install solartherm using Ubuntu 18.04 and then 20.04, but the download link for the solstice and dakota tar.gz files hosted at "cloudstor.aarnet.edu.au" is not working in both cases.

I would like to know if you could attach the tar.gz files here, or store them in the solartherm git, so I can complete the installation since when doing scons and scons install it fails because it cannot find solstice and dakota.

Thank you very much in advance.

kind regards!

Xiang.

jdpipe, 2024-03-22:
Hi Xiang
Thanks for the notification. This 'cloudstor' service was recently taken offline. I'll work to migrate the files to another storage and update our instructions. When we've got it sorted, I'll update you here.
Cheers
JP

yewang726, 2024-05-05:
Hi Xiang,

We have updated the links in the installation instruction. Please have a try and let me know if you still meet any issues.

Cheers,
Ye

closed 2024-05-22 by yewang726