SolarTherm/SolarTherm #118: st_simulate issues on Linux Mint 22.1

arfontalvoANU, 2025-03-12:
Hi all,

I installed OpenModelica and SolarTherm on a virtual machine with Linux Mint 22.1 using these steps:

```
sudo apt-get update
sudo apt-get install ca-certificates curl gnupg
sudo curl -fsSL http://build.openmodelica.org/apt/openmodelica.asc | \
  sudo gpg --dearmor -o /usr/share/keyrings/openmodelica-keyring.gpg

echo "deb [arch=amd64 signed-by=/usr/share/keyrings/openmodelica-keyring.gpg] \
  https://build.openmodelica.org/apt \
  $(cat /etc/os-release | grep "\(UBUNTU\\|DEBIAN\\|VERSION\)_CODENAME" | sort | cut -d= -f 2 | head -1) \
  release" | sudo tee /etc/apt/sources.list.d/openmodelica.list

sudo apt update
sudo apt install openmodelica

echo 'installPackage(Modelica,"3.2.3");' > ~/install_modelica.mos
omc ~/install_modelica.mos
rm ~/install_modelica.mos

sudo apt install python3-pip
sudo apt install python3-testresources

python3 -m pip install --upgrade pip
python3 -m pip install --upgrade setuptools wheel
python3 -m pip install scons scipy matplotlib DyMat pyswarm cma pytest openpyxl distro colorama

python3 -m pip install solsticepy 
sudo apt install libpolyclipping-dev libtbb-dev libyaml-dev  libgomp1
cd ~
wget https://www.meso-star.com/projects/solstice/downloads/Solstice-0.9.1-GNU-Linux64.tar.gz
tar -zxvf ~/Solstice-0.9.1-GNU-Linux64.tar.gz 
export PATH=$PATH:~/Solstice-0.9.1-GNU-Linux64/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/Solstice-0.9.1-GNU-Linux64/lib

sudo apt install openmpi-bin openmpi-common openmpi-doc libopenmpi-dev
sudo apt install git

cd ~
git clone https://github.com/SolarTherm/SolarTherm.git SolarTherm
cd ~/SolarTherm
scons
scons install
```

Then, I tested `st_simulate` and got the following error:
```
Compiling model
Error processing file: /home/arfontalvo/SolarTherm/examples/SimpleSystem.mo
Failed to parse file: /home/arfontalvo/SolarTherm/examples/SimpleSystem.mo!

Error: Failed to insert class SimpleSystem within examples;

 the available classes were:

# Error encountered! Exiting...
# Please check the error message and the flags.
Failed to parse file: /home/arfontalvo/SolarTherm/examples/SimpleSystem.mo!

Execution failed!
Traceback (most recent call last):
  File "/home/arfontalvo/.local/bin/st_simulate", line 175, in <module>
    sim.compile_model()
  File "/home/arfontalvo/.local/lib/python3.12/site-packages/solartherm/simulation.py", line 318, in compile_model
    sp_run(call)
  File "/usr/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['omc', '-s', '-q', '-n=0', '-d=nonewInst', '-i=SimpleSystem', '/home/arfontalvo/SolarTherm/examples/SimpleSystem.mo', 'Modelica', 'SolarTherm']' returned non-zero exit status 1.
[... 90 lines cut ...]
|                 | |       | | LOG_NLS            [logging for nonlinear systems]
|                 | |       | | LOG_NLS_V          [verbose logging of nonlinear systems]
|                 | |       | | LOG_NLS_HOMOTOPY   [logging of homotopy solver for nonlinear systems]
|                 | |       | | LOG_NLS_JAC        [outputs the jacobian of nonlinear systems]
|                 | |       | | LOG_NLS_JAC_TEST   [tests the analytical jacobian of nonlinear systems]
|                 | |       | | LOG_NLS_NEWTON_DIAG [Log Newton diagnostic]
|                 | |       | | LOG_NLS_RES        [outputs every evaluation of the residual function]
|                 | |       | | LOG_NLS_EXTRAPOLATE [outputs debug information about extrapolate process]
|                 | |       | | LOG_RES_INIT       [outputs residuals of the initialization]
|                 | |       | | LOG_RT             [additional information regarding real-time processes]
|                 | |       | | LOG_SIMULATION     [additional information about simulation process]
|                 | |       | | LOG_SOLVER         [additional information about solver process]
|                 | |       | | LOG_SOLVER_V       [verbose information about the integration process]
|                 | |       | | LOG_SOLVER_CONTEXT [context information during the solver process]
|                 | |       | | LOG_SOTI           [final solution of the initialization]
|                 | |       | | LOG_SPATIALDISTR   [logging of internal operations for spatialDistribution]
|                 | |       | | LOG_STATS          [additional statistics about timer/events/solver]
|                 | |       | | LOG_STATS_V        [additional statistics for LOG_STATS]
|                 | |       | | LOG_SUCCESS        [this stream is always active, unless deactivated with -lv=-LOG_SUCCESS]
|                 | |       | | LOG_SYNCHRONOUS    [log clocks and sub-clocks for synchronous features]
|                 | |       | | LOG_ZEROCROSSINGS  [additional information about the zerocrossings]
LOG_ASSERT        | debug   | unrecognized option -lv -LOG_SUCCESS,-stdout
Execution failed!
Simulation time: 0.067285s
```

jdpipe, 2025-03-19:
the default setting in st_simulate seems to be to turn off LOG_SUCCESS notification since we already capture and report non-success via other routes (I think) and turning of stdout (has that been replaced by LOG_STDOUT?) presumably because of the noisy output of many solvers. But I wonder if suppressing all that solver output is a good thing for the default behaviour...?

jdpipe, 2025-03-19:
It seems we must either remove these 'suppressions' (maximises support for different omc vesions) or else update them for new omc command syntax (keeps output as similar as possible to how it was in the past)

jdpipe, 2025-03-19:
As discussed, would be great if we can develop a GHA build for this new ubuntu version as well as for much more recent omc. See .github/actions/main.yml